utkabobr/SliceBeam #1: No Kobra 3 profile

joelpr1234, 2024-11-12:
I'm curious, you have a Klipper firmware for Kobra 3, but this app doesn't have a profile for the Kobra 3? Also I don't see any profiles for Flashforge 5M/pro.

Is there a way to add them? Like transferring a folder or a file from Bambu slicer to Orca slicer?

What printer are you testing this app?

utkabobr, 2024-11-12:
Pre-configured profiles are from PrusaSlicer's repo, it might be missing new printers

joelpr1234, 2024-11-13:
ok, I'll try to export from Orca, then import to SliceBeam. I'll will reply back. Thanks.

joelpr1234, 2024-11-13:
Where I get a .ini file from Orca? I get json files only.

utkabobr, 2024-11-13:
You need an .orca_printer file

joelpr1234, 2024-11-13:
> You need an .orca_printer file

Ok, thanks. I got it now. I got 3 printers imported, 2 K3 combos and the 5M Pro.

Print tab, Not working is the first import print configs, still showing custom profile and all setting need to be changed like layers and speed.

Filament tab, setting are incomplete, I see at least max volumetric speed transferred the setting, but not the temperatures.

Printer tab, same, it transferred some settings but not others, for example, it did not transfer bed shape, max print height or retraction length. All very important. 

How I get all of them imported? I see it has 3 imports, Print Configs, Filaments and Printers, just the first I don't get anything to choose when I use my .orca_printer files.

By the way, I can't donate using Boosty, do you have anything else for donations?

Thank you.

utkabobr, 2024-11-13:
> just the first I don't get anything to choose when I use my .orca_printer files
It means that it failed to import Print settings profiles, Orca is ughhh... strange thing, many things have changed in their format without a visible reason for me, I've made a converter work only on provided to me printer profiles.

Can you please attach your orca_printer file?

utkabobr, 2024-11-13:
> By the way, I can't donate using Boosty, do you have anything else for donations?

I can only open a crypto wallet as an alternative, direct transfers or visa/mastercard payments aren't available in my country ;(

utkabobr, 2024-11-13:
By the way, here is my Kobra 3 profile (rename .txt to .ini), based off settings from built-in gcodes and tuned a bit for my preferences:
[SliceBeam_config_bundle.txt](https://github.com/user-attachments/files/17737260/SliceBeam_config_bundle.txt)

utkabobr, 2024-11-21:
> [orca_printer files.zip](https://github.com/user-attachments/files/17737912/orca_printer.files.zip)
> 
> Here are my 3 printers settings in Orca.
> 
> Thanks for the response. I wish you dedicate a little more time to the kobra 3 full klipper firmware, you're a genius, I don't know anything about coding.

I've checked out your profiles. All three contains only filament & printer profiles, no print settings. It's exported like that

joelpr1234, 2024-11-23:
> I've checked out your profiles. All three contains only filament & printer profiles, no print settings. It's exported like that

Then it's an Orca exportation problem, I tried to import to other oca based software and it tells me this: There are 0 configs imported. (Only non-system and compatible configs)
Hint: Make sure you have added the corresponding printer before importing the configs.

So I think Orca is not exporting print settings? or what I'm doing wrong when I try to export?