utkabobr/SliceBeam #2: Orcaslicer profile import failed.

guitarplayer0171, 2024-11-20:
[MyKlipper 0.4 nozzle - Copy for real this time1.zip](https://github.com/user-attachments/files/17837118/MyKlipper.0.4.nozzle.-.Copy.for.real.this.time1.zip)
When I try to import my Orca profile from my desktop, I get the error "java.lang.StringIndexOutOfBoundsException: begin 17, end -1, length 26"

Felix14-v2, 2024-11-21:
AFAIK Orca is not supported, only PrusaSlicer.

utkabobr, 2024-11-21:
> AFAIK Orca is not supported, only PrusaSlicer.

Orca profiles is supported in experimental state via internal profile converter

utkabobr, 2024-11-21:
> [MyKlipper 0.4 nozzle - Copy for real this time1.zip](https://github.com/user-attachments/files/17837118/MyKlipper.0.4.nozzle.-.Copy.for.real.this.time1.zip) When I try to import my Orca profile [...]

I'll tune a bit error handling, but the real problem is that you're trying to import child profile without parent

guitarplayer0171, 2024-11-21:
Is there something I am doing wrong in exporting the profile? I tried earlier with the bare .json profile that's in the orca appdata folder, but Slice Beam complained about it not containing any .ini files. which is when I did the export profile thing in orca to get the .orca_printer file. Trying the exported process presets (what I thought would be the "parent" you were talking about) didn't help either, it was another .json file that Slice Beam said didn't have a .ini in it.

guitarplayer0171, 2025-01-25:
I'll add that I've since gotten it more or less working with a PrusaSlicer profile I put together to work with my printer setup. Other than some slight UI things I need to get used to, it seems to work pretty decently. Two things that don't work and I don't even know if they're possible features are that I can't see what layer I'm on in mainsail since it's not putting the proper layer information in on each layer, that could be something I'm doing wrong, though. The other thing is that there's no gcode thumbnail generation, which I get is pretty unnecessary, so I guess that's something I'm not too worried about.

closed 2025-01-25 by guitarplayer0171