utkabobr/SliceBeam #7: Wrong printer model and print start sequence

T4psi, 2025-03-16:
I sliced a model for the Prusa Mini (with input shaper). When I started the print, it warned me that the printer model is incorrect. I thought that maybe it's just some glitch and started the print anyway. It started right6away, without mesh leveling and on 230 °C for PLA. I had to stop it immediately because I sas that the nozzle left a mark on the bed. 

[image]

T4psi, 2025-03-16:
Gcode: https://drive.google.com/file/d/1ikFqU5K2jdmPBr4MRwIuPftv4UE4GCog/view?usp=drivesdk

utkabobr, 2025-03-16:
> Gcode: https://drive.google.com/file/d/1ikFqU5K2jdmPBr4MRwIuPftv4UE4GCog/view?usp=drivesdk

This link has closed access. Can you please zip gcode and attach here or open the access?

T4psi, 2025-03-16:
I've updated the access rights.