utkabobr/SliceBeam #12: Custom start gcode

furrywall, 2025-05-26:
The custom start geode is not being adhered to. It always uses the default. Am I missing something here?

[image]

[image]

chick911t, 2025-05-30:
Same problem my self, haven't been able to remove the preset start gcode it just dose both,

furrywall, 2025-05-30:
I solved it by installing the apk under latest release instead of the play store.

chick911t, 2025-05-30:
I recently downloaded the latest akp and was still unable to get rid of the standard purge line and start gcode, it always dose a 2 full bed mesh the long purge line in the y direction and my active purge line I will down load the akp again and give it a go

chick911t, 2025-05-31:
[image]

furrywall, 2025-05-31:
Also try to use a custom profile when first setting up the app. I was choosing ender 3v2 profile but I downloaded the apk and used a custom profile. Try that maybe?